p3d-dev/Print-To-3D #21: Exported gcode doesn’t print

DavidLozzi, 2024-11-17:
using this on my ipad
importing a stl (tried with two different files, both do the same)
slice it
export to gcode, the file has size, 1.4mb, other file 8.7mb
using SD card bring to my Ender 3 V3 SE
Files are listed to print, but opening shows 0 lines, time, etc.
print it anyway it appears to go through the motion but no filament is used, nothing prints

[image]
[image]

DavidLozzi, 2024-11-17:
I think I misspoke, it doesn’t go through the motions, it moves some and then stops.

p3d-dev, 2025-12-17:
stale issue